AllenCellModeling/aicsimageio #520: `Dimensions` typing

multimeric, 2023-07-05:
## System and Software
* aicsimageio Version: `4.7.0`
* Python Version: `3.9.16`
* Operating System: MacOS 12.6

## Description

When accessing fields of a `Dimensions` object, I tend to get the following error with my type checker, which is Pyright:
```
Cannot access member "T" for type "Dimensions"
  Member "T" is unknownPylance[reportGeneralTypeIssues]
```

This makes sense, because the fields are only ever `setattr`'d onto the class: https://github.com/AllenCellModeling/aicsimageio/blob/7e537c8ea650dda6c3cf3e3d2b53eb54fad3fd7f/aicsimageio/dimensions.py#L109.

## Expected Behavior
I expect `dimensions.T` to not cause a type checker error

## Reproduction
```python
from aicsimageio.dimensions import Dimensions
def fun(dims: Dimensions):
    return dims.T
```

toloudis, 2023-09-26:
I think we try to get around this by having custom __setattr__ and __getattr__ functions in place. I wonder why Pyright doesn't like that.

multimeric, 2023-09-27:
Yep, I can't reproduce this with newer `aicsimageio`, I think it was fixed in [v4.9.1](https://github.com/AllenCellModeling/aicsimageio/releases/tag/v4.9.1)+ and I was using 4.7.X.

closed 2023-09-27 by multimeric