AllenCellModeling/aicsimageio #523: tifffile min version incorrect, and now incompatible with bfio extra

tlambert03, 2023-07-16:
#487 uses the `is_mmstack` kwarg to `TiffFile`.  That feature was not available until version `2023.3.15`, so the pin in setup.py needs to be updated to `"tifffile>=2023.3.15"`

however, that now becomes internally incompatible with the bfio extra which states `tifffile<2022.4.22` ... I'm not sure why that's there however, (@evamaxfield added it in #397 so it could perhaps be removed at this point)

All that said though, given the ubiquity of tifffile in the ecosystem, I actually think you'd be better off trying to maintain support for older versions of tifffile in order to be an "easier" library to install with other libraries.  so I think you might consider rethinking how you solved #487, or at least check for the kwarg and allow it to fail in older versions of tifffile

tlambert03, 2023-07-16:
cc @BrianWhitneyAI on #487

multimeric, 2023-08-08:
This is a pretty major issue for a project I'm working on. It seems that `bfio` itself pins `tifffile>=2021.8.30,<2022.4.22`: https://github.com/PolusAI/bfio/blob/fc52266d35fbf22ba44dd4f258200b1415ff9448/setup.py#L37, so it's not as simple as removing our pin, I think first `bfio` needs to be fixed to be compatible with the changed `tifffile` API, then we can remove `aicsimageio`'s pin on `tifffile` and add a new minimum `bfio` version, and _then_ downstream users such as me can add a minimum `aicsimageio` version and everything should work.

SeanLeRoy, 2023-09-19:
@tlambert03 @toloudis I found _a_ solution by reverting the `tifffile` stuff back, it seems older versions of `tifffile` have issues with newer versions of `zarr` so that would have to be restricted in combination. See #529.

toloudis, 2023-09-19:
It looks like there is a PR on bfio waiting for review, which could help all of this:
https://github.com/PolusAI/bfio/pull/48/files#diff-60f61ab7a8d1910d86d9fda2261620314edcae5894d5aaa236b821c7256badd7

SeanLeRoy, 2023-09-26:
Closing assuming #529 provided an adequate solution. Feel free to re-open if that isn't the case

closed 2023-09-26 by SeanLeRoy