AllenCellModeling/aicsimageio #524: ome_metadata fro czi file

TomPJWyatt, 2023-07-19:
## System and Software
* aicsimageio Version: 4.11.0
* Python Version: 3.11.4
* Operating System: mac os 13.3

## Description
I am trying to access the ome_metadata of a czi file.
I.e. I am doing:
```
from aicsimageio import AICSImage
img = AICSImage('test.czi')
img.ome_metadata
```
But I get a long error...

ValidationError: 36 validation errors for OME
instruments -> 0 -> light_source_group -> 0 -> id
  string does not match regex "(urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:LightSource:\S+)|(LightSource:\S+)" (type=value_error.str.regex; pattern=(urn:lsid:([\w\-\.]+\.[\w\-\.]+)+:LightSource:\S+)|(LightSource:\S+))
.
.
.

SeanLeRoy, 2023-07-20:
Could you provide an example image so I can see if I am able to reproduce this?

evamaxfield, 2023-07-20:
If I remember correctly from prior similar issues, this may be that the value is simply `0` and not `LightSource:0`.

Which should be easy to fix. I think we had portions of the code which already do that for other attributes in the `_fix_ome_metadata` or something. Might also be a fix in XSLT though.

TomPJWyatt, 2023-07-21:
Hi thanks for the help both! Here's a link to an image that produces the error for me: https://we.tl/t-0pioFes0qb

TomPJWyatt, 2023-08-01:
Hi @SeanLeRoy did you manage to reproduce this with the data I uploaded? Many thanks

SeanLeRoy, 2023-08-01:
Hi @TomPJWyatt I haven't had a chance to look at this and the link to the file expired, I'll set some reminders to check this today and tomorrow if you have a chance to set up a new link, thanks!

SeanLeRoy, 2023-08-03:
Thank you for the updated link, got it downloaded and will take a look and report back

adizhol-gp, 2023-08-10:
+1

**pydantic_core._pydantic_core.ValidationError: 1 validation error for Laser wavelength
  Input should be a valid number, unable to parse string as a number [type=float_parsing, input_value='', input_type=str]
    For further information visit https://errors.pydantic.dev/2.1/v/float_parsing**

```

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/aicsimageio/readers/czi_reader.py", line 890, in ome_metadata
    return metadata_utils.transform_metadata_with_xslt(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/aicsimageio/metadata/utils.py", line 95, in transform_metadata_with_xslt
    ome = OME.from_xml(str(ome_etree))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/ome_types/_mixins/_base_type.py", line 201, in from_xml
    return cast(T, from_xml(xml, **kwargs))
                   ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/ome_types/_conversion.py", line 134, in from_xml
    return parser.parse(xml_2016, OME_type)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/bases.py", line 54, in parse
    result = handler.parse(source)
             ^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/handlers/lxml.py", line 52, in parse
    return self.process_context(ctx)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/handlers/lxml.py", line 67, in process_context
    self.parser.end(
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/bases.py", line 152, in end
    return item.bind(qname, text, tail, objects)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/nodes/element.py", line 88, in bind
    obj = self.config.class_factory(self.meta.clazz, params)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/config.py", line 10, in default_class_factory
    return cls(**params)  # type: ignore
           ^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/ome_types/_mixins/_kinded.py", line 22, in __init__
    return super().__init__(**data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/ome_types/_mixins/_base_type.py", line 117, in __init__
    super().__init__(**data)
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/pydantic/main.py", line 159, in __init__
    __pydantic_self__.__pydantic_validator__.validate_python(data, self_instance=__pydantic_self__)
pydantic_core._pydantic_core.ValidationError: 1 validation error for Laser
wavelength
  Input should be a valid number, unable to parse string as a number [type=float_parsing, input_value='', input_type=str]
    For further information visit https://errors.pydantic.dev/2.1/v/float_parsing
ome_xml = czi.ome_metadata
/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/converter.py:85: ConverterWarning: Failed to convert value `` to one of (<class 'float'>,)
  warnings.warn(
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/aicsimageio/readers/czi_reader.py", line 890, in ome_metadata
    return metadata_utils.transform_metadata_with_xslt(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/aicsimageio/metadata/utils.py", line 95, in transform_metadata_with_xslt
    ome = OME.from_xml(str(ome_etree))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/ome_types/_mixins/_base_type.py", line 201, in from_xml
    return cast(T, from_xml(xml, **kwargs))
                   ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/ome_types/_conversion.py", line 134, in from_xml
    return parser.parse(xml_2016, OME_type)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/bases.py", line 54, in parse
    result = handler.parse(source)
             ^^^^^^^^^^^^^^^^^^^^^
[... 1 lines cut ...]
    return self.process_context(ctx)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/handlers/lxml.py", line 67, in process_context
    self.parser.end(
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/bases.py", line 152, in end
    return item.bind(qname, text, tail, objects)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/xsdata/formats/dataclass/parsers/nodes/element.py", line 88, in bind
    obj = self.config.class_factory(self.meta.clazz, params)
[... cut ...]
  File "/home/adiz/miniconda3/envs/gpdev/lib/python3.11/site-packages/pydantic/main.py", line 159, in __init__
    __pydantic_self__.__pydantic_validator__.validate_python(data, self_instance=__pydantic_self__)
pydantic_core._pydantic_core.ValidationError: 1 validation error for Laser
wavelength
  Input should be a valid number, unable to parse string as a number [type=float_parsing, input_value='', input_type=str]
    For further information visit https://errors.pydantic.dev/2.1/v/float_parsing
```

SeanLeRoy, 2023-08-28:
Hello @TomPJWyatt, I am able to reproduce this error. It appears that `aicsimageio` is converting the file's metadata from:
```
<LightSource Id="MTBLKM980LaserLine561">
    <Power>13</Power>
    <LightSourceType>
        <Laser>
            <Tuneable>false</Tuneable>
        </Laser>
    </LightSourceType>
</LightSource>
```
to
```
<ome:Laser ID="MTBLKM980LaserLine561" Wavelength=""/
```
because the xslt here:
```
<xsl:when test="LightSourceType/Laser">
   <xsl:element name="ome:Laser">
         <xsl:attribute name="ID">
             <xsl:value-of select="@Id"/>
         </xsl:attribute>
         <xsl:attribute name="Wavelength">
             <xsl:value-of select="LightSourceType/Laser/Wavelength"/>
         </xsl:attribute>
   </xsl:element>
</xsl:when>
```
is taking the LightSource tag's ID attribute and assigning it to the Laser ID's attribute.

This becomes a problem since "MTBLKM980LaserLine561" is not a number/float. We could do something like what @evamaxfield is describing to catch and modify the attribute prior to trying to convert the metadata into an OME object, but we would run into the issue @adizhol-gp may have also encountered where the wavelength is not set even though it is expected. I am no CZI/OME metadata expert so the best course of action here is not clear to me. It seems we could do one of the following:
* Catch and try to convert string-like LightSource IDs into Laser IDs & assign an arbitrary(?) number to the wavelength
* Ignore LightSource/Laser when the wavelength is not set
* Someone double checks my logic and finds something up with my logic or a cooler way to do this
* Ignore this case altogether?

SeanLeRoy, 2023-11-08:
Closing due to inactivity. `aicsimageio` is moving into "maintenance" mode where only high impact bugfixes (or community contributed) work will be done in `aicsimageio`. Instead of `aicsimageio`, we are creating a package soon to be released called bioio. [See the reason for this change here](https://github.com/AllenCellModeling/aicsimageio/discussions/372). 

If this issue is still relevant to anyone (@TomPJWyatt) feel free to re-open this issue and I'll move it over to the repository that will be responsible for CZI reads ([bioio-czi](https://github.com/bioio-devs/bioio-czi)) or feel free to open the issue over there yourself.

closed 2023-11-08 by SeanLeRoy