AllenCellModeling/aicsimageio #539: Specify resolution level for reading OME-Zarr

tischi, 2023-11-07:
## Use Case

For processing OME-Zarr image data it will be very useful to specify with resolution level to work on.

## Solution

Currently the resolution level appears to be [hard-coded here](https://github.com/AllenCellModeling/aicsimageio/blob/main/aicsimageio/readers/ome_zarr_reader.py#L71).

We could see two ways to implement this:

1. Specify the resolution level `R` during the opening of the file
2. Have `R` as an additional in the ACISImage dimensions

The second option seems more flexible, but probably is less easy to implement in your current framework?

toloudis, 2023-11-07:
This is something we are very likely to add to the API in [bioio](https://github.com/bioio-devs/bioio) which is getting close to its first release.  

What do you think of setting the resolution level in the same way that we handle scenes?  Something like `get_num_resolution_levels` and then `set_current_resolution_level`.  And then you can call `.dims` and `.get_image_data` as usual.

SeanLeRoy, 2023-11-08:
I like the solution you proposed @toloudis , would the setter you describe raise `NotYetImplemented` for readers/plugins by default? We would of course immediately add an implementation for the OME ZARR reader.

toloudis, 2023-11-09:
Should we take this thread into the bioio ticket?
I'd say that the setter function should raise a ValueException if you pass a value that is out of range of `range(get_num_multiresolution_levels)`.   So basically, it should only accept 0 for anything that doesn't have multiresolution data.

joshmoore, 2023-11-09:
> [tischi](https://github.com/tischi) commented [2 days ago](https://github.com/AllenCellModeling/aicsimageio/issues/539#issuecomment-1799578587)
> maybe can express a more informed opinion on how to exactly implement this?!

If in doubt:

https://github.com/ome/bioformats/blob/develop/components/formats-api/src/loci/formats/IPyramidHandler.java#L53

SeanLeRoy, 2023-12-01:
AICSImageIO is entering maintenance mode meaning we will only be applying easy-to-implement critical bugfixes. Replacing AICSImageIO is BioIO (soon to be officially released) with the same features just refactored for various reasons seen [here](https://bioio-devs.github.io/bioio/MIGRATION.html).

[Here is a similar issue in BioIO](https://github.com/bioio-devs/bioio-base/issues/8), but feel free to open another issue in BioIO or add onto that one if you feel it doesn't capture your request.

closed 2023-12-01 by SeanLeRoy