AllenCellModeling/aicsimageio #553: Questions regarding compression method choice with  OmeTiffWriter.save()

AlanWu886, 2024-02-16:
Hi,

Thank you all for developing this wonderful package!
I am not quite sure if my question is actually a bug report or feature request so please allow me to start with some background and questions.

## Use Case
We were trying to convert some .lif files to .tif without specifying any compression method and assumed there should not be a drastic change of image size by doing so. However, we found out the image was 5 times smaller than the original .lif file (from 500MBs to 100MBs).
Then I tried to figure the cause in the ome_tiff_writer.py. And in line 304 i see:
compression=**TIFF.COMPRESSION.ADOBE_DEFLATE**,

I just want to ask if there is any reason to use adobe deflate as the default? Will I encounter any unwanted behavior by setting this to None? Is there any way for us to specify the compression method without changing the original package code right now?

## Solution
If the answer to last question above is no, could be have a parameter in the save() function letting us choose the compression method in the future?
Maybe something like save(....., **compress=<compression method>**)?

Thanks a lot for your time reading this post!
Alan

toloudis, 2024-02-16:
Hi Alan,

First of all, deflate compression is lossless so you should not have to worry about the integrity of the actual pixel data.

The main reason to choose compression by default is that for large scale bioimaging data, storage gets expensive fast.  And decompression is relatively fast compared to data transmission time.

Unfortunately, in order to change the compression method we would have to modify the code to add this as a parameter.  However, you can feel free to copy it to a new writer file and make changes yourself.  It definitely makes sense to add this as a parameter.

As for changing the value to None, I am not sure what the `tifffile` library does with that, off the top of my head.  We are just passing the value along to that wonderful library.

AlanWu886, 2024-02-16:
Hi @toloudis 

Thanks for your timely response!

Just FYI, after the conversion with deflate compression we tried to examine the output images and found out some values were actually changed out of some weird reason. I will verify this again with my supervisor.  

I understand aicsimageio is phasing into maintenance mode and it might not make sense for you to have this feature on it. 
Would you please consider having this feature on bioio if you think others can also benefit from this while this is not causing too much trouble for you or other contributors? We are developing tools based on aicsimage/bioio for the whole department and I feel it will be easier for distribution purpose if the change is implemented on your end in my humble opinion.

Again, thanks a lot for the tips and helps!
Alan

closed 2025-03-02 by github-actions[bot]