AllenCellModeling/aicsimageio #554: Importing fails

onurserce, 2024-02-17:
## System and Software
* aicsimageio Version: 4.14.0
* Python Version: 3.12.2
* Operating System: Apple M1 Max Sonomo 14.1 (23B74)

## Description
`from aicsimageio import AICSImage` produces an error (`import aicsimageio` doesn't work as well).

`(DVP) ➜  ~ python
Python 3.12.2 | packaged by conda-forge | (main, Feb 16 2024, 20:54:21) [Clang 16.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from aicsimageio import AICSImage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/onur_serce/anaconda3/envs/DVP/lib/python3.12/site-packages/aicsimageio/__init__.py", line 6, in <module>
    from .aics_image import AICSImage  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/onur_serce/anaconda3/envs/DVP/lib/python3.12/site-packages/aicsimageio/aics_image.py", line 12, in <module>
    from ome_types import OME
  File "/Users/onur_serce/anaconda3/envs/DVP/lib/python3.12/site-packages/ome_types/__init__.py", line 17, in <module>
    from ome_types._conversion import from_tiff, from_xml, to_dict, to_xml, validate_xml
  File "/Users/onur_serce/anaconda3/envs/DVP/lib/python3.12/site-packages/ome_types/_conversion.py", line 16, in <module>
    from xsdata_pydantic_basemodel.bindings import (
  File "/Users/onur_serce/anaconda3/envs/DVP/lib/python3.12/site-packages/xsdata_pydantic_basemodel/bindings.py", line 32, in <module>
    class SerializerConfig(config.SerializerConfig):
  File "/Users/onur_serce/anaconda3/envs/DVP/lib/python3.12/site-packages/xsdata_pydantic_basemodel/bindings.py", line 35, in SerializerConfig
    *config.SerializerConfig.__slots__,
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'SerializerConfig' has no attribute '__slots__'. Did you mean: '__class__'?`

## Expected Behavior
*What did you expect to happen instead?*
Import without an error message

## Reproduction
*A minimal example that exhibits the behavior.*

1. Create a conda environment via `conda env create -f DVP.yaml`
2. Activate it via `conda activate DVP`
3. Initiate Python and import aicsimageio

## Environment
*Any additional information about your environment*
- Conda environment file: https://www.dropbox.com/scl/fi/u20esvwb944n2rbl072g1/DVP.yaml?rlkey=356spbpoezywk25vwda2yrj69&dl=0

onurserce, 2024-02-17:
Hi,

I have made some further tests, just to be able to get this running.

- I have pinned the python to`python=3.9` (installs 3.9.18. I have also tried pinning to 3.8) by editing the YAML file and created the conda environment again. Both versions however still produced the same error.
- I have created another conda environment with `python=3.9` (again, actually installs 3.9.18). With this, aicsimageio import works.

So it seems like there is a conflict with other packages. Any ideas how this can be fixed or how I can narrow down on the issue?

tlambert03, 2024-02-19:
this is an issue caused by the release of xsdata v24.2 yesterday, which broke ome-types, which depends on xsdata.  The problem isn't related to this repo; it's been reported and fixed in ome-types: https://github.com/tlambert03/ome-types/issues/242

I will push a new release later today, but the quick fix for you is to downgrade xsdata with `pip install xsdata<24.2`

going to close this since it's not an aicsimageio issue, but feel free to ask more questions

closed 2024-02-19 by tlambert03