AllenCellModeling/aicsimageio #555: tifffile dependency: version pin outdated?

m-albert, 2024-03-04:
Hi, thanks a lot for your work which makes dealing with all the different file formats so much easier!

I wanted to report a dependency issue here:

I'm running into issues trying to install `aicsimageio[all]` together with newer versions of `tifffile`. Specifically,

`pip install "tifffile>2022.7.28" "aicsimageio[all]==4.14.0"` would yield

> ERROR: Cannot install aicsimageio, aicsimageio[all]==4.14.0 and tifffile>2022.7.28 because these package versions have conflicting dependencies.

> The conflict is caused by:
    The user requested tifffile>2022.7.28
    aicsimageio 4.14.0 depends on tifffile<2023.3.15 and >=2021.8.30
    aicsimageio[all] 4.14.0 depends on tifffile<2023.3.15 and >=2021.8.30
    aicsimageio[all] 4.14.0 depends on tifffile<2022.4.22; extra == "all"

I digged a bit into this and found that the tifffile pin in aicsimageio is attached to the `bfio` dependency and placed here:

https://github.com/AllenCellModeling/aicsimageio/blob/e5197d1073c046ea87d694070da8d0299d840bf4/setup.py#L39

However, `bfio` has different tifffile version requirements that are compatible with newer tifffile versions:

https://github.com/PolusAI/bfio/blob/v2.3.0/setup.py

Could it be that the tifffile pin by aicsimageio is outdated? Or is there maybe a reason for the tifffile pin that's independent of the bfio dependency with which it is listed together?

toloudis, 2024-03-06:
Thanks for writing here.  It's certainly possible that the tifffile pin is outdated.  I wonder if you can write a pull request that unpins it and all tests still pass?  We have seen incompatibilities with tifffile vs aicsimageio before, related to micromanager support.  Also please consider trying out `bioio` which is the successor to aicsimageio.  It uses a plugin model for the file format readers, so it's possible that you'd only have this conflict if you installed the bfio plugin.

m-albert, 2024-03-06:
Cool, I wasn't aware that `bioio` is a more modular successor of `aicsimageio`. That's good to know and I'll try it out soon.

I'll open a PR here unpinning `tifffile` as you suggested.

Thanks for your help @toloudis.

m-albert, 2024-03-06:
Just found there's already a PR with the exact same change https://github.com/AllenCellModeling/aicsimageio/pull/547.

The PR seems to already be approved! I commented there that it should solve this issue.

m-albert, 2024-03-07:
Thanks @toloudis . Closing as #547 is merged.

closed 2024-03-07 by m-albert